fragmuffin/pygcode #5: Line number in program

TurBoss, 2017-08-20:
Hello,

I'm trying to process a program that contains N codes at the begining of each line

while trying to process the block in the machine
an error is raised 

```
self.machine.process_block(line.block)
```

```
/usr/bin/python2.7 /home/turboss/Sources/hazzy/hazzy/modules/kremlin/kremlin.py
Traceback (most recent call last):
  File "/home/turboss/Sources/hazzy/hazzy/modules/kremlin/kremlin.py", line 395, in <module>
    main()
  File "/home/turboss/Sources/hazzy/hazzy/modules/kremlin/kremlin.py", line 386, in main
    kremlin.draw_path()
  File "/home/turboss/Sources/hazzy/hazzy/modules/kremlin/kremlin.py", line 332, in draw_path
    self.add_points(line, points)
  File "/home/turboss/Sources/hazzy/hazzy/modules/kremlin/kremlin.py", line 370, in add_points
    self.machine.process_block(line.block)
  File "/home/turboss/Sources/hazzy/hazzy/modules/pygcode/machine.py", line 471, in process_block
    self.process_gcodes(*block.gcodes, modal_params=block.modal_params)
  File "/home/turboss/Sources/hazzy/hazzy/modules/pygcode/machine.py", line 452, in process_gcodes
    modal_gcode = self.modal_gcode(modal_params)
  File "/home/turboss/Sources/hazzy/hazzy/modules/pygcode/machine.py", line 422, in modal_gcode
    ' '.join(str(x) for x in unasigned_words), self.mode
hazzy.modules.pygcode.exceptions.MachineInvalidState: modal parameters 'N10' cannot be assigned when in mode: <Mode: G00 G17 G90 G91.1 G94 G21 G40 G49 G54 G61 G97 M05 M09 F0 S0 T0>
N10 (HAZZY SPLASH G-CODE)
```

I was trying to figure what is wrong but failed

Thanks

fragmuffin, 2017-08-22:
@TurBoss thanks for the issue!

My CAM software doesn't generate line numbers, so I hadn't picked this problem up yet.

**Workaround**

define the following class (anywhere)

    class GCodeLineNumber(pygcode.gcodes.GCode):
        @classmethod
        def word_matches(cls, w):
            return w.letter == 'N'
        word_letter = 'N'
        word_value_configurable = True
        exec_order = 0

**If you're interested**

The issue is that the parser pulls all words (`<letter><number>`) out of the gcode string, then attempts to bind them as parameters to the closest valid gcode to the left of it.
Anything that remains (in your case, `N10`) is considered to be a parameter to a modal motion gcode (eg: `G0`)...
So what this is saying is that `G0 N10` isn't a valid code, which it isn't... but that's because I haven't assigned `N` codes to be their own dummy gcode.

I'll be releasing v`0.1.3` very soon to fix this, and some other issues.

fragmuffin, 2017-08-22:
_correction_: make that v `0.2.0`, I'll move this lib to `alpha`

fragmuffin, 2017-08-22:
ok, [new version](https://github.com/fragmuffin/pygcode/releases/tag/0.2.0) is out... let me know if you have any more troubles.

closed 2017-08-22 by fragmuffin

TurBoss, 2017-08-22:
Thank yo so much