fragmuffin/pygcode #12: Support for variables (#3), math (#3*5+#<gcb>) and Brackets ( X[56*#<scale>] )

maculata, 2018-04-11:
This appears to be missing.... did I miss something?  Thanks

fragmuffin, 2018-04-11:
@maculata 
No, you're not missing anything, the current version of `pygcode` doesn't support this.

Do you have some sample gcode you could post here?
Also: is there an online specification you can link.

I'd imagine this would be done by pre-processing the gcode, which may be out of scope of `pygcode`, but I'll look into it.

fragmuffin, 2018-04-11:
linuxcnc specs:

* [parameters](http://linuxcnc.org/docs/html/gcode/overview.html#_parameters)
* [expressions](http://linuxcnc.org/docs/html/gcode/overview.html#gcode:expressions)

maculata, 2018-04-12:
(Mill - generated: Sat Apr  7 22:26:21 2018 )
(Version: 2.0.2)
(Description = engrave02)
(Material = Aluminum : -any-)

(Units = G20 inches)
(Work Offset = G54)
(Tool Number = 101)
(Tool Description = Drag Knife)
(Tool Diameter = 0.0010 inches)
(Spindle RPM = 0)
(Text to engrave = 0123)
(Font = FONT.TTF)
(Font height = 0.5000)
(Scale = 0.000117564072419)
(X Base Location = 0.0000)
(Justification Setting = left)
(Y Base Location = 1.3500)
(Feed Rate = 20.0 inches/minute)

(Z Clear Location = 0.0800)
(Z Start Location = 0.0250)

(Z Depth of Cut = 0.0300)
(Ammend Serial Number = no)

(----- Start of G-code -----)
(<cv1>)

G17 G90  (XY Plane, Absolute Distance Mode)

G64 P 0.0050 Q 0.0000 (Path Blending)

(font: FONT.TTF)
(text: 0123)
#1=0.055000  (SafeHeight)
#2=0.030000  (Depth of Cut)
#3=0.000118  (XY Scale)
#4=20.000000  (Feed)
#5=0.000000  (X offset)
#6=1.350000  (Y offset)
#7=0.025000  (Z offset)
(Fill Line Scale = 24)
(Fill = 0)

G17 (set current plane to XY)
G90 (set Absolute Distance Mode)
G20 (set units to inches)
G54 (set current work offset)

G30 (move to preset G30)
T101 M6 G43 H101
M5 (0 RPM. turn off spindle)
M8 (set coolant to ON)

G00 X [#5] Y [#6]
G00 Z [#1+#7]
(start of symbol 0)
(starting X offset: 0)
G00 Z [#1+#7]
G00 X [1393*#3+#5] Y [2077*#3+#6] (moveto)
G01 Z [0-#2+#7] F#4
G5.1 X[1577*#3+#5] Y[1716*#3+#6] I[139*#3] J[-156*#3]
G5.1 X[1597*#3+#5] Y[1159*#3+#6] I[20*#3] J[-90*#3]
G5.1 X[1602*#3+#5] Y[991*#3+#6] I[0*#3] J[-69*#3]
G5.1 X[1602*#3+#5] Y[815*#3+#6] I[4*#3] J[-90*#3]
G5.1 X[1585*#3+#5] Y[655*#3+#6] I[-5*#3] J[-106*#3]
G5.1 X[1307*#3+#5] Y[180*#3+#6] I[-69*#3] J[-299*#3]
G5.1 X[795*#3+#5] Y[4*#3+#6] I[-209*#3] J[-176*#3]
G5.1 X[313*#3+#5] Y[176*#3+#6] I[-271*#3] J[0*#3]
[... 126 lines cut ...]
G5.1 X[5973*#3+#5] Y[762*#3+#6] I[-135*#3] J[45*#3]
G5.1 X[5854*#3+#5] Y[893*#3+#6] I[-107*#3] J[53*#3]
G5.1 X[5973*#3+#5] Y[1081*#3+#6] I[-21*#3] J[123*#3]
G5.1 X[6206*#3+#5] Y[1167*#3+#6] I[118*#3] J[41*#3]
G5.1 X[6411*#3+#5] Y[1540*#3+#6] I[205*#3] J[115*#3]
G5.1 X[6218*#3+#5] Y[1917*#3+#6] I[0*#3] J[238*#3]
G5.1 X[5985*#3+#5] Y[1991*#3+#6] I[-102*#3] J[74*#3]
G5.1 X[5706*#3+#5] Y[1892*#3+#6] I[-160*#3] J[0*#3]
G5.1 X[5518*#3+#5] Y[1737*#3+#6] I[-61*#3] J[-61*#3]
G5.1 X[5456*#3+#5] Y[1724*#3+#6] I[-33*#3] J[-13*#3]
G5.1 X[5356*#3+#5] Y[1763*#3+#6] I[-57*#3] J[0*#3]
G5.1 X[5313*#3+#5] Y[1860*#3+#6] I[-43*#3] J[39*#3]
G5.1 X[5420*#3+#5] Y[2064*#3+#6] I[0*#3] J[86*#3]
G5.1 X[5604*#3+#5] Y[2212*#3+#6] I[90*#3] J[103*#3]
G5.1 X[5870*#3+#5] Y[2306*#3+#6] I[164*#3] J[82*#3]
G5.1 X[6206*#3+#5] Y[2277*#3+#6] I[156*#3] J[21*#3]

G0 Z [#1+#7]

M9 (stop coolant)
M5 (stop spindle)

G30 Z [#1+#7] (move in Z only to preset G30)
G30 (move to preset G30)
M30 (end program)

fragmuffin, 2018-04-12:
@maculata 
That looks quite simple, you're not using any dynamic variables like `#<_rpm>`, so you can pre-process the file to remove parameters.

I think this does the job:

```python
#!/usr/bin/env python

import sys
import re
filename = sys.argv[1]

regex_paramset = re.compile(r'^#(?P<name>\w+)=(?P<value>[^\s\(]+)')
regex_expression = re.compile(r'\[(?P<exp>[^\]]+)\]')
regex_param = re.compile(r'#(?P<name>\w+)')

params = {}

with open(filename, 'r') as fh:
    for line in fh:
        # parameter set
        match = regex_paramset.search(line)
        if match:
            params[match.group('name')] = float(match.group('value'))
            sys.stdout.write("(%s)\n" % line.rstrip('\n'))
            continue # don't print line
        
        # evaluate expressions (with parameter substitution)
        for match in reversed(list(regex_expression.finditer(line))):
            exp_str = match.group('exp')
            for m in reversed(list(regex_param.finditer(exp_str))):
                exp_str = exp_str[:m.start()] + ("%g" % params[m.group('name')]) + exp_str[m.end():]
            exp_result = eval(exp_str)
            line = line[:match.start()] + ("%g" % exp_result) + line[match.end():]

        sys.stdout.write(line)
```

so if the content you've pasted above is in a file called `input.g`, we can create a pre-processed file called `output.g` with:

```bash
./pre-process.py input.g > output.g
```

I haven't tested it, and it has limits, but it should do the trick for now as a workaround.

fragmuffin, 2018-04-13:
To clarify:
I agree with @maculata ; parameters should be deciphered by `pygcode` interpreter where possible, and shouldn't _need_ pre-processing.

Furthermore, the pre-processing done by the above script is something `pygcode` should do, namely a new feature for [`pygcode-norm`](https://github.com/fragmuffin/pygcode/wiki/Script-pygcode-norm)

I'm currently planning dialects for `pygcode`, variables and expressions will be a part of that.

### Design
stuff to think about, mostly brain-storming

**Variables in different Dialects**

* Each dialect may have a different set of pre-defined parameters (such as `#<_rpm>`, `#<_x>`)
* Parameter values belong to the virtual machine.

**Expressions in different Dialects**

* Different sytnax : could other dialects use `{}` brackets as opposed to `[]`, (TODO: research)
* Support switchable : do some dialects not support praameters, syntax may conflict with another dialect feature.

**Line Processor Idea**

* Deciphering parameters & expressions may be the role of a "line processing" class instance
    * each dialect may employ said processor, an altered processor, or none at all.
    * multiple processors per interpreter are in an ordered list (allowing different dialects to prioritise differently)
    * existing line decoding may be moved to this design, expression processing pre-processed

Ideas welcome

BillyBumbler, 2018-04-18:
Good day folks!

This is quite the conundrum.... And is why G-code files are not universal and cannot be shared across most machines. I program CNC machines in an industrial manufacturing environment, and this was very disappointing when I first learned it. Every machine controller can have different capabilities and can choose to interpret g-code files in it's own way. This is why your CAM software of choice requires a "Post" file to help it generate a G-code file that is compatible with your machine. Many times, the only way to utilize some features of the controller is to **_hand edit_** the G-code file yourself (or write your own plug-in for your CAM software). The machine I use at work uses an Osai 10 Series controller which interprets code differently than a Haas machine would, for example. It's easy for a hobbyist to assume that G-Code files are all interchangeable because most of them use the same software (Linux CNC or Mach 3) in their controllers, but this is not the case in the rest of the CNC world.

Basically, the "dialects" you speak of will be a configuration file of sorts to help pygcode emulate certain controllers, is that right? Or perhaps you could choose to emulate a LINUX CNC or MACH 3 based machine with the dialects being different post file configurations? I look forward to see how this goes!