fragmuffin/pygcode #18: M117 string parameter

dickshaydle, 2019-05-24:
Hi there, 

i just discovered pygcode and this seems to be a really useful project.

But i have some problems using it with marlin. It doesn't this seem to be easily possible to add a command with arbitrary string parameters. This is needed for commands like M117 which is used to display text on an LCD display.
I tried the workaround from [#9](https://github.com/fragmuffin/pygcode/issues/9) but it only takes "Words"/ one letter Arguments.

Did i miss something?

fragmuffin, 2019-05-26:
Hi @dickshaydle 
you're absolutely right, `pygcode` is a bit limited, and more of a pet project (hence it's alpha release state).
The completion of #9 and #14 would greatly help you, but it's a little hard for me to add to my workload.

To clarify, I believe you've tried something like (but not exactly):

```python
>>> import pygcode
>>> class LCDisplayGCode(pygcode.GCode):
...     word_key = pygcode.Word('M', 117)
>>> from pygcode import text2gcodes
>>> text2gcodes('M117 blah blah')
[...]
GCodeWordStrError: word 'B' value invalid
```

Which makes sense... because the `LCDisplayGCode` isn't configured to have parameters, so it assumes "blah blah" is just more encoded gcodes.

I'm sorry to say that there isn't an easy fix for this, there's an assumption built into `dialects` that limits this.

For example, the `O` GCode sets the name of the project, which extends all the way to the end of the line, this works because of the `value_regex`:
https://github.com/fragmuffin/pygcode/blob/5577d6560ebb50c3eff4806caa30bf965737e124/src/pygcode/dialects/linuxcnc.py#L134-L139

However, `M` codes are limited to the numbers following the `M`
https://github.com/fragmuffin/pygcode/blob/5577d6560ebb50c3eff4806caa30bf965737e124/src/pygcode/dialects/linuxcnc.py#L120-L125

As your example has indicated, this is not only wrong, it's a fundamental limit of the library as it stands.

I would love to fix this, but I'm very short on time of late... is there any urgency on your side to get this running?
I have some ideas of how you can get around this, but they're very hacky.

dickshaydle, 2019-05-29:
Hi @fragmuffin,
thanks a lot for your answer. 

This is not time critical for me. I would also want to add something to the project and just wanted to ask what the current status is, to not reinvent the wheel.

So as it seems to me, it would be reasonable to first implement the dialects and afterwards change the word class or add a new 'line' class or so.

fragmuffin, 2019-05-30:
@dickshaydle 

> I would also want to add something to the project and just wanted to ask what the current status is

I guess the project's status is, _stale_... it's not exactly active; as it hasn't had a change or a release in many months, but I'd be happy to accept pull requests.

> So as it seems to me, it would be reasonable to first implement the dialects and afterwards change the word class or add a new 'line' class or so.

Dialects will need to be implemented in a few stages, I'll tag you in a comment in #9

petaflot, 2024-03-08:
see https://github.com/petaflot/pygcode