fragmuffin/pygcode #34: PrusaSlicer gcode file compatibility

aissak21, 2022-05-03:
The following code chunk worked well for a gcode file sliced in CURA but not in PrusaSlicer. Now I get `error: word 'P' value invalid`. Is there a way to go about this to parse this file with pygcode, perhaps if it's not a compatibility issue? There is a PyGcode extension for Prusa but it's for adding code lines into the file via python. 

`
from pygcode import Line

gcode_lines = []
gcode_block = []
gcode_params = []

with open(test_file2, 'r') as test:
  for line_text in test.readlines():
      line = Line(line_text) # decode gcode line
      line.block.gcodes  # list of gcodes
      line.block.modal_params  # motion modal parameters
      
      gcode_lines.append(line)
      gcode_block.append(line.block.gcodes)
      gcode_params.append(line.block.modal_params)

print(gcode_block, sep='\n')
`

xlxs4, 2022-11-28:
In my file, I've encountered:
`word 'P' value invalid`, solved by commenting `M862.3 P "MK3S" ; printer model check`
`block code remaining '.1'`, solved by commenting `M115 U3.10.1 ; tell printer latest fw version`
`word 'W' value invalid`, solved by commenting `G28 W ; home all without mesh bed level`

rongith, 2023-04-03:
Check there: https://github.com/emorybarlow/pygcode